Snapmaker/SnapmakerController-IDEX #1: Feature Request: Enable filament runout sensor when printing from Octoprint

evilC, 2023-05-18:
# 🐞 feature request

### Is your feature request related to a problem? Please describe.
While printing from Octoprint, the filament runout sensor is not supported

### Describe the solution you'd like?
Enable filament runout sensor
I think this just requires `HOST_PROMPT_SUPPORT` to be enabled in `Configuration_adv.h` ?

747lulu747, 2023-05-23:
We are now doing compatibility with octoprint. It take some time. Please keep an eye out for commits on github

TylonHH, 2023-06-03:
just uncommend in the firmware
`#define HOST_ACTION_COMMANDS`
see refernce

Googliola, 2023-09-05:
> We are now doing compatibility with octoprint. It take some time. Please keep an eye out for commits on github

No commits since initial release. Issues get ignored for months. What's the plan?

Googliola, 2023-09-08:
HOST_ACTION_COMMANDS enabled. Does not change a thing.

TylonHH, 2023-09-08:
> HOST_ACTION_COMMANDS enabled. Does not change a thing.
How is it enabled? Do I need a special firmware? Because OctoPrint says still disabled with the latest official firmware

Googliola, 2023-09-08:
No, I recompiled FW with HOST_ACTION_COMMANDS  enabled. Here's the binary (remove .txt):
[J1_V2.2.11_20230905.bin.txt](https://github.com/Snapmaker/SnapmakerController-IDEX/files/12558877/J1_V2.2.11_20230905.bin.txt)

evilC, 2023-11-03:
I spoke to a Marlin dev about this, and enabling `HOST_ACTION_COMMANDS` will not fix it. Apparently SM rewrote part of the filament sensor code in Marlin or something. IIRC he said that because of this, some code will need to be changed in the UI, which we do not have access to, so we will be unable to fix it ourselves.

TylonHH, 2023-11-03:
So that's why I also can’t use M2000 via USB.