Snapmaker/SnapmakerController-IDEX #2: Feature Request: Filament loading position is not illuminated

evilC, 2023-05-18:
# 🐞 feature request
When you use the filament load/unload feature, the position it moves to is at the very front of the printer.  
This area is not covered well by the lights in the printer.  

### Describe the solution you'd like?
Suggest moving this position to around the middle on the Y axis, so that you can see what you are doing better.   
This is especially useful, for example, if you are trying to load filament and want to check if the filament is going through the filament path properly, and also to check the state of the gears

747lulu747, 2023-05-24:
Thank you, this is a good proposal and we will change it after evaluating testing.

evilC, 2023-07-31:
I should also add that the X position needs changing too. It's too close to the silicone pads,. so it's not easy to clean the nozzle while it is in the load position.  
Furthermore, there needs to be a "Park" option ion the load/unload page to move the extruder back over the silicone pad once you are done loading.
Also, ideally, while loading / unloading, the X/Y steppers should ideally be disabled so that you can move the print heads around if you need to. In fact, if you did this, then you would not need to make any of the above changes really, as we could easily position the heads where we wanted them