Snapmaker/SnapmakerController-IDEX #3: Bug: When performing "Heated bed levelling", J1 waits for right extruder to heat up, even though it is not used

evilC, 2023-05-18:
# 🐞 bug report
When you enable the "Heated bed levelling" function, the J1 waits for the right extruder to come up to temperature, even though it is not used for the calibration

### Affected Version(s)
All

### Is this a regression? (optional)
No

Expected behavior
When you enable the "Heated bed levelling" function, it should only wait for the bed and extruder 1 to heat up, as that is all that is used in the function

747lulu747, 2023-05-23:
It is indeed possible to close the right extruder when Heated bed leveling. But, when doing calibration, after the heated bed leveling, it is the Z offset calibration, which needs the right extruder been heated up. Therefore, we choose to heat the right extruder when the heated bed leveling when the left extruder is heating. This isn't a bug, it's a design choice.

evilC, 2023-05-23:
> It is indeed possible to close the right extruder when Heated bed leveling

Are you saying that it is possible to cancel heating of right extruder when doing bed levelling and not wait?

747lulu747, 2023-05-24:
Yes. But we need to change the source code and update the firmware.