Snapmaker/SnapmakerController-IDEX #4: Feature Request: Load filament does not work while octoprint connected.

evilC, 2023-05-18:
# 🐞 feature request
When connected to Octoprint, the "Load filament" option activated via the touchscreen does not work

### Is your feature request related to a problem? Please describe.
Sometimes (but not always it seems), when I am connected to the J1 via Octoprint, the "Load Filament" option does not work.  
The UI updates and says it is loading, but the gears do not turn.

### Describe the solution you'd like?
Filament load function should work while connected to Octoprint

747lulu747, 2023-05-23:
Hi evilC, thanks for your feedback. Indeed, the "Load Filament" function do not work in some situation. It is bugs of screen application. We will fix it in next updation.

Googliola, 2023-09-05:
> It is bugs of screen application. We will fix it in next updation.

Hmmm, no more updates then for J1. Product sold and abandoned?

Christalwater83, 2023-11-06:
Hi,
Write in Octoprint Console first M18 than M17.
Now Load/Unload should work.

M18 = Steppers Off
M17 = Steppers On