Snapmaker/SnapmakerController-IDEX #6: Feature Request: Detail recovery procedure

evilC, 2023-05-18:
# 🐞 feature request
I am tempted to start trying to compile the firmware with various fixes, but I am concerned that should something not work, I will have problems restoring to stock firmware.  

The wiki says `Download the firmware from the Snapmaker website > Insert the USB flash drive into the printer > Turn on the printer > Tap the blank area on the home screen of the touchscreen > Tap USB > Tap the firmware file to update`  

- What happens if the LCD is inoperable? Normally with printers, if it sees a firmware file on the USB stick, it will upgrade. Is this the case for the J1?  
- I do not see where on the Snapmaker website where the stock firmwares are available for download

### Describe the solution you'd like?
There Should be a reliable way to restore the J1 to a stock firmware

747lulu747, 2023-05-26:
Hi evilC, 

> What happens if the LCD is inoperable? Normally with printers, if it sees a firmware file on the USB stick, it will upgrade. Is this the case for the J1?
No, it is not. The updating of firmware right now require the screen APP . The USB interface is connected to the screen, not to the Marlin. So, It can not update firmware without screen.
J1 controller can be upgraded through the serial port. However, an understanding of the upgrade protocol and a dedicated updating software tool are required. In fact, we do have a dedicated software upgrade tool, but this tool only supports Simplified language. We are developing an simple upgrade tool that supports English. At that time, we will post on the J1 official forum, titled such as "How to upgrade through the serial port" on the J1 official forum.

747lulu747, 2023-05-29:
Here is the guidance of upgrading J1 frimware via USB without any screen operation: https://forum.snapmaker.com/t/firmware-upgrade-via-serial-port/31566

evilC, 2023-05-29:
Thank you, but how would we go back to stock firmware? I am guessing that we could use a stock compiled binary, but I do not see them for download anywhere? Should the stock binaries not be hosted on GitHub?

747lulu747, 2023-05-31:
Hi, evilC, please refer to this link: https://github.com/Snapmaker/SnapmakerController-IDEX/releases.
The [J1_MC_APP_V2.2.11_20230530.bin](https://github.com/Snapmaker/SnapmakerController-IDEX/releases/download/v2.2.11/J1_MC_APP_V2.2.11_20230530.bin) could be use as a stock firmware.

evilC, 2023-05-31:
I take it that this is the original firmware which came with the J1 when it first shipped?  
I notice that there does indeed seem to be a place to download all firmwares ([On the Snapmaker site](https://forum.snapmaker.com/t/snapmaker-j1-firmware-updates-and-downloads/29972?fbclid=IwAR0Fp5jzHy8xps5AsNUhLGnteSe7hOmZCfY8FNhkvmHsKjN2rRBc2IE-VJY)), so this poses a few questions...  

1. Will this repo be updated at some point to contain the changes in the current release (2.5.13)?  
2. Are the versions on the Snapmaker site compatible with the serial port upload method? ie can we flash 2.5.13 via serial port using the Python updater tool?

KyGunsAndRadio, 2023-06-01:
I've been wondering the same thing with regards to the version differences. The builds on the download page begin at v2.3.5 with current being 2.5.13. The repo version is currently 2.2.11, but there is a tag point within the main branch labeled as 2.5.11. I'm totally confused as to what version this repo actually contains right now.

[image]

747lulu747, 2023-06-20:
> 

@KyGunsAndRadio Hi, We're sorry for the confusion the version caused you.
We mainly have two versions, one is the packaged version of firmware + screen, and the other is the firmware version on this github. The v2.5.13 on the download page is the version number of the firmware + screen. And the version of firmware is based on the latest tag of our github, which is currently v2.2.11.