Snapmaker/SnapmakerController-IDEX #10: Layer Shifting

jledgett, 2023-05-22:
This is an already known issue, but posting here for tracking purposes.

The J1 encounters an unexpectedly high number of failed prints due to layer shift issues. 

I have attempted to print larger models using both a dual extruder (Polymaker PC-FR in left extruder, PolyDisolve S2 in the right extruder) and single extruder (PC-FR in left extruder).  In almost all of the attempts, these larger prints suffer layer shifts and result in a useless print.  

Belts have already been tensioned using the belt tensioning tools printed with the Snapmaker-provided PLA.   A decent number of other users are reporting this issue too, so not likely a unit-specific issue.

I have uploaded two example GCODE files, evidence photos, along with the associated log files to this link:
https://drive.google.com/drive/folders/1ZvvkiYfUxH5ddxh6Y5LxWpnU2P1t6BhH?usp=share_link

I have also included my current Cura profile at this link also.

Googliola, 2023-05-29:
I keep getting multiple layer shifts too. No octoprint attached.

The likelihood of layer shifts increases with Z. I suspect that the under-powered stepper motor is getting hot after a while and causes missed steps. 

Enclosure doors closed with PETG
[image]

jledgett, 2023-05-29:
@Googliola are you using single or dual extruder?

Googliola, 2023-05-29:
Single extruder (left)

jledgett, 2023-05-30:
Are yours on the X or Y axis or both.  I think from the photos it looks like the X?

Googliola, 2023-05-30:
That is correct. Only X axis so far.
Why are you asking?

jledgett, 2023-05-30:
Struggling with layer shifts myself, but mine was on primarily on the Y axis

747lulu747, 2023-05-31:
Many reasone can cause a layer shifts. Here are some of the reasons and responses we summarized：

1. Warping cause by Insufficient heat dissipation or high printing speed and warping will increase the friction of printing and product a layer shift. Slow down the printing speed improve heat dissipation may fix it.
2. High friction of XY linear rails. Turn off the power of the printer and slide the print head by hand to see if it is smooth.
3. Bug in firmware that will cause discontinuous steps when print large modle without G92 E0 (to reset the E position) in gcode.  This bug has been fixed in version V2.2.10

jledgett, 2023-05-31:
@747lulu747 No warping, running latest firmware, no obvious friction on the rails.  

What is the motor amperage rating for the Y stepper.   You have it driving 1400ma in firmware.  Unless it os a 2.0A or higher peak stepper setup, then you may be overdriving the motor which can cause missed steps.

Googliola, 2023-05-31:
@747lulu747 No warping either, downgraded to 2.5.10, almost 0 friction

> You have it driving 1400ma in firmware. Unless it os a 2.0A or higher peak stepper setup, then you may be overdriving the motor which can cause missed steps.
That's a good point. The Nema17 seem a bit short to provide that kind of peak power? Maybe less amperage is better? (If only the documented process of firmware compilation would actually produce a useable .bin file....)
But I suspect that the culprit is to be found elsewhere: OVERHEATING. Given the size of the motor(s), the enclosure, and the fact that it happens mostly on large prints, I try the following:
1. Slightly reduced tension on the X-axis in my case to reduce motor stress. [That already seemed to help with other large prints](https://github.com/Snapmaker/SnapmakerController-IDEX/issues/10#issuecomment-1567043174)
2. Ordered heatsinks for the X motors, will add one to Y-motor too if space allows.
3. Remove top lid. Not my favourite option as it makes enclosure futile.

jledgett, 2023-05-31:
If they are TMC2209 drivers, the maximum suggested run current for them is 1.2A, so 1.4A is overdriving it for sure!

On prints with repeated movement on the Y axis this could be overheating the stepper driver — especially if you already have ambient heat build-up due to something like a long-running print, a warm printing environment (i.e. a garage or light industrial space), or printing with higher temps (PA or PC).

@747lulu747 can you confirm what drivers and motors are used?

Googliola, 2023-05-31:
TMC2209 continuous drive current is 2A and peak current is 2.8A. (according to bigtreetech GitHub)

In the J1 firmware config, TMC2209 are configured. So I guess that's the case

jledgett, 2023-05-31:
But you are correct: Trinamicms product page indicates 2.0A RMS.  Not sure where Voron got their number from.

Googliola, 2023-06-01:
> > TMC2209 continuous drive current is 2A and peak current is 2.8A. (according to bigtreetech GitHub)
> 
> Interesting, Voron’s driver current documentation stated lower: https://docs.vorondesign.com/community/howto/120decibell/calculating_driver_current.html

Hmm, that is weird. I must admit though, that I suck at physics / electronics, so I do not really understand what and why is being claimed or even in what ways the two statements relate to each other (Where does that .707 factor come in again? Read about it elsewhere but, well, I never understood).

Long story short, my gut tells me the drivers are not the main culprits causing the layer shifts. It's underpowered motors and heat-tolerance.

747lulu747, 2023-06-20:
> If they are TMC2209 drivers, the maximum suggested run current for them is 1.2A, so 1.4A is overdriving it for sure!
>
> On prints with repeated movement on the Y axis this could be overheating the [...]

> If they are TMC2209 drivers, the maximum suggested run current for them is 1.2A, so 1.4A is overdriving it for sure!
>
> On prints with repeated movement on the Y axis this could be overheating the [...]

Our motor is customized. The Y-axis utilizes a 42-inch hybrid stepper motor with a length of 48. The X-axis has a length of 40.