Snapmaker/SnapmakerController-IDEX #17: Bug: PID Hotend PID setting not possible on a per-extruder basis

evilC, 2023-05-24:
# 🐞 bug report

### Affected Version(s)
All

### Is this a regression? (optional)
No

### To Reproduce
Modify PID settings for E0
```
Send: M301 E0 P15.24 I0.96 D60.28
Recv: echo: p:15.24 i:0.96 d:60.28
Recv: ok
```

Modify PID settings for E1
```
Send: M301 E1 P15.56 I0.99 D60.95
Recv: echo: p:15.56 i:0.99 d:60.95
Recv: ok
```
Save to EEPROM
```
Send: M500
Recv: echo:Settings Stored (734 bytes; crc 22982)
Recv: ok
```

### Exception or Error (optional)
Retreive settings
```
Send: M503
[...]
Recv: echo:  M301 P15.56 I0.99 D60.95
Recv: echo:  M301 P15.56 I0.99 D60.95
```

(Note same values listed twice)

### Expected behavior
Response should be something like:
```
Recv: echo:  M301 E0 P15.24 I0.96 D60.28
Recv: echo:  M301 E1 P15.56 I0.99 D60.95
```
I spoke to someone in Marlin Discord channel about this and we think this is an underlying issue in the version of Marlin used, and not a result of any modifications that Snapmaker has done.

747lulu747, 2023-05-26:
Hi evilC, thank for your proposal. Current version do not enable PID_PARAMS_PER_HOTEND featrue. We will consider adding this feature in the next version.

evilC, 2023-07-05:
> Hi evilC, thank for your proposal. Current version do not enable PID_PARAMS_PER_HOTEND featrue. We will consider adding this feature in the next version.

It looks like that without this feature enabled, the right extruder is not using the PIDs **at all**: https://github.com/MarlinFirmware/Marlin/issues/24644

> When PID_PARAMS_PER_HOTEND is disabled, the PID constants that are supposed to be shared between all extruders will only work for the first extruder. For any other extruders, the PID algorithm receives zero for all constants. This causes PID mode to always output a power of zero, resulting in the temperature oscillating around the PID_FUNCTIONAL_RANGE threshold as if it were a bang-bang controller.