Snapmaker/SnapmakerController-IDEX #18: Feature Request: Add Z-offset adjustment while printing via Octoprint

evilC, 2023-05-24:
### Is your feature request related to a problem? Please describe.
Z-Offset tweaking (AKA "BabyStepping") is only available while a job is running that was started via the printer touchscreen.  

### Describe the solution you'd like?
This feature should be available while running a job via Octoprint

747lulu747, 2023-05-26:
We currently only support adjusting Z_offset via screen. We are planning to be more compatible with cotoprint, and we will add this feature at that time. In the next week or two, we will firstly list what feature should we add in next version and give release time.

747lulu747, 2023-06-20:
Hi evilC, We are not compatible with octoprint within 2 weeks, very sorry, hope you can understand.  Some things take time, and the time of compatibility of octoprint time is uncertain.

evilC, 2023-08-24:
As a side note until this gets fixed, the workaround is to issue an `M206` command