Snapmaker/SnapmakerController-IDEX #19: Feature request: enable use of ntc thermistor

switch87, 2023-05-25:
For some (I suspect it's a good one) reason the j1 has ptc thermistors by default. But they are not available aftermarket. Now I destroyed one of the original ones and ordered spare parts on the european webshop... That are paid but are not getting shipped. I have ntc thermistors but I can't use them. Since this is a firmware thing I think it would be nice to be able to switch to ntc so the printer can be used with easy to get aftermarket parts.

747lulu747, 2023-05-26:
Hi switch87. This is a good suggestion. We can consider using the M command to switch between NTC and PTC or other thermistors. We will make a decision after fully considering the requirements and the impact of changes. If you need to use the NTC you received immediately, you can send us your NTC model and parameters. We can compile a special firmware for you. However, this may also take a while.

switch87, 2023-05-26:
Thank you for your reply @747lulu747 .
The thermister is an "E3D Prusa Thermistor", I don't have more data on it. But since many printers are based on prusa, I think it's very standard?
I only have it in my left extruder, the right extruder still haar the factory one from jadelabo/snapmaker.