Snapmaker/SnapmakerController-IDEX #22: Bug: when printing via USB connection (backside) and go into touchscreen "control" and back the temperature turns off

TylonHH, 2023-06-03:
### Affected Version(s)
FW 2.5.13

### To Reproduce
1. print via USB connection (eg OctoPrint)
2. touch on display at the printer "control" (Steuerung)
3. press back
4. check the temperature, its dropping

### Expected behavior
- no effect when fine tuning via display

KyGunsAndRadio, 2023-06-12:
This behavior is also exhibited even without OctoPrint. Simply trying to preheat using the touch screen isn't really viable because as soon as you leave the "Control" screen, all the temps are reset to zero. So by the time you find your model and begin printing, everything has started to cool down.

747lulu747, 2023-06-20:
Hi, @KyGunsAndRadio @TylonHH . The J1 sreen will reset the nozzle to 0 after leaving the control page. It is mainly for safety reasons, preventing users from forgetting to turn off the heating after setting a temperature. This is a design strategy.

icu-hobby, 2023-11-11:
> strategy.

Frankly, it is possible check if the printer is working (example: motors are moving during printing) and reset temperatures only if printer is not working after some time (for preheat nozzles/bed before printing).